balena-labs-projects/balena-sense #7: Temperature too high, humidity too low

jackhistoria, 2019-04-23:
Hi, thanks for your cool project. I got that little dude up and running.
Sadly the temperature is about 2-15°C too high and the humidity is 10% too low
compared to a synthetic hygro- and thermometer. The Air quality is always around 50%
doesn't matter if the window is open or not.
I ordered that sensor: https://www.aliexpress.com/item/BME680-Digital-Temperature-Humidity-Pressure-Sensor-CJMCU-680-High-Altitude-Sensor-Module-Development-Board/32961416338.html?spm=a2g0s.9042311.0.0.7e364c4dss6GLZ
and soldered 3.3V, ground, scl and sda properly to the pi zero w.

I left SD0 and CS out at the breakout - just 4 connected like described here:https://www.balena.io/blog/build-an-environment-and-air-quality-monitor-with-raspberry-pi/?utm_source=efp&utm_medium=etcher&utm_campaign=balena-sense&utm_content=v3

Did I do something wrong or is the breakout sensor defect or is there some adjustment needed and if yes, where?

Thanks in advance,

Jack

MrStrangerCid, 2019-04-25:
Afaik, you need to wire it in 5V just like I did in my project. Sensor reading is -+2 margins.
I did purchase the same bme680 just like yours.

chrisys, 2019-04-25:
I found my temperature/humidity readings to be acceptable out of the box, but you can define offsets within `sensor/scripts/bme680_air_quality.py`. Around line 19 you can add sensor.set_temp_offset(-3.5), for example. The sensor was affected quite a lot if I had it in close proximity to the Pi.

A good way to test the air quality reading is by using some kind of spirits/alcohol or paint thinner, or simply a glass of wine! Putting the sensor anywhere near these liquids should give you a vast difference.

jackhistoria, 2019-04-29:
> Afaik, you need to wire it in 5V just like I did in my project. Sensor reading is -+2 margins.
Thanks for your recommendation, sadly no effect.

> I found my temperature/humidity readings to be acceptable out of the box, but you can define offsets within `sensor/scripts/bme680_air_quality.py`. Around line 19 you can add `sensor.set_temp_offset(-3.5)`, for example. The sensor was affected quite a lot if I had it in close proximity to the Pi.
Thanks for your recommendation, sadly also no effect, not even a little relative change :-(
 
> A good way to test the air quality reading is by using some kind of spirits/alcohol or paint thinner, or simply a glass of wine! Putting the sensor anywhere near these liquids should give you a vast difference.
I will do that later, thanks for that tip.

Anyways, i am sure I got a bad production - got a money return from ali express and will buy simply another one.  maybe I solder it on a longer wire or try some other python project with that one.
Would have been cool to just be able to adjust the offset - anyways - ill 
[image]
get back here in about a month ;-)

chrisys, 2019-05-27:
@jackhistoria thanks for the update! The air quality reading is always relative, so ideally you need to start the application with good air quality. That being said I made a change last week as we were polling the sensor every 1 second during the warmup phase and then every 10 seconds whilst it was running. This allowed the sensor to cool which was affecting the reading. I've changed it to continue taking readings every 1 second but store every 10, but I'm also looking at using the Bosch libraries for the sensor to see what difference that makes.

chrisys, 2019-06-04:
@jackhistoria just to let you know we've now updated this project with the Bosch library usage and offsets for temperature and humidity, and altitude correction for pressure too - details in the README! I'd love to get your thoughts when your replacement sensor arrives.

jackhistoria, 2019-06-05:
OMG Thats incredible!! Thanks for that so much!! 
Maybe I got both sensors working with that when it arrives, you are awsome.
Will get back after I tested it.

chrisys, 2019-07-15:
Closing this as the offsetting features were added in v1.4.0.

closed 2019-07-15 by chrisys