balena-labs-projects/balena-sense #8: install RTIMULib prebuilt wheel or apt package?

shaunmulligan, 2019-04-24:
It would increase the build speed quiet a bit (on local push mostly) if we installed from
https://www.piwheels.org/project/RTIMULib/ or via `apt-get install python3-rtimulib`

chrisys, 2019-04-25:
@shaunmulligan I always encounter serious issues with this particular package, related discussion here: https://github.com/RPi-Distro/python-sense-hat/issues/58

I'd rather remove it entirely if we could, I don't think it's even used, but is a dependency of the Sense-HAT libs.

drewmccal, 2019-05-17:
Yeah, I was curious why this was included.  Poking around the files, I didn't see any reference to RTMU being used.

> @shaunmulligan I always encounter serious issues with this particular package, related discussion here: [RPi-Distro/python-sense-hat#58](https://github.com/RPi-Distro/python-sense-hat/issues/58)
> 
> I'd rather remove it entirely if we could, I don't think it's even used, but is a dependency of the Sense-HAT libs.

chrisys, 2019-05-21:
This is resolved in https://github.com/balena-io-projects/balena-sense/pull/10

The downside of this is that for users with a Sense-HAT, they'll no longer get a pressure reading, but for everyone else using the BME680 the image is lighter and builds much faster.

closed 2019-05-21 by chrisys