balena-labs-projects/balena-sense #22: Sensor container permissions error

mkeley1, 2019-06-26:
Hi,

Having an issue deploying this project to a Pi0.

Upon pushing the project to my device running BalenaOS the "Sensor" container that is created is in a constant boot loop and the logs are showing the below error
`26.06.19 14:39:22 (+0800)  sensor  /bin/sh: 1: ./entry.sh: Permission denied`

When trying to SSH into the device via the WebUI being told the Container is restarting and when trying to SSH via the BalenaCLI` exec failed: cannot exec a container that has stopped: unknown`

Have tried both Production and Development environments.

HostOS: balenaOS 2.32.0+rev1
Supervisor: 9.14.0

Sakatard, 2019-06-26:
Hey, this was fixed in the most recent commit please close.

mkeley1, 2019-06-26:
Have since found the changes committed last-night.

closed 2019-06-26 by mkeley1

chrisys, 2019-06-26:
@mkeley1 thanks for reporting and glad it was resolved for you with the latest version. Can you confirm what OS you're using on your computer to push the app (Windows/macOS/Linux)?

mkeley1, 2019-06-26:
I'm pushing from a Windows 10 machine.