balena-labs-projects/balena-sense #25: Balena push failing due to missing download

Razzeee, 2019-07-04:
When I run `balena push <name>` it errors when it tries to access this URL that seems to be unavailable https://ae-bst.resource.bosch.com/media/_tech/media/bsec/BSEC_1.4.7.3_Generic_Release_20190410.zip

```
[Info]      Building on arm01
[Info]      Pulling previous images for caching purposes...
[Success]   Successfully pulled cache images
[sensor]    Step 1/16 : FROM balenalib/raspberrypi3-python:3-build
[telegraf]  Step 1/7 : FROM balenalib/raspberrypi3
[influxdb]  Step 1/4 : FROM balenalib/raspberrypi3-alpine
[grafana]   Step 1/8 : FROM balenalib/raspberrypi3
[influxdb]   ---> e76d3be3914b
[influxdb]  Step 2/4 : RUN apk add influxdb
[influxdb]  Using cache
[influxdb]   ---> 91bbe9d0b526
[influxdb]  Step 3/4 : RUN sed -i 's|/var/lib/influxdb|/data/influxdb|g' /etc/influxdb/influxdb.conf
[grafana]    ---> 9a857048cc34
[grafana]   Step 2/8 : COPY ./grafana.ini /usr/share/grafana/conf/custom.ini
[telegraf]   ---> 9a857048cc34
[telegraf]  Step 2/7 : RUN curl -o /tmp/telegraf.deb https://dl.influxdata.com/telegraf/releases/telegraf_1.11.0-1_armhf.deb
[influxdb]  Using cache
[influxdb]   ---> 78d1fb50cdef
[influxdb]  Step 4/4 : CMD influxd
[influxdb]  Using cache
[influxdb]   ---> af0d701ab067
[influxdb]  Successfully built af0d701ab067
[telegraf]  Using cache
[telegraf]   ---> 71d494b966b0
[telegraf]  Step 3/7 : RUN dpkg -i /tmp/telegraf.deb && rm /tmp/telegraf.deb
[grafana]   Using cache
[grafana]    ---> f482ae92f502
[grafana]   Step 3/8 : COPY ./provisioning /usr/src/app/provisioning
[telegraf]  Using cache
[telegraf]   ---> ed2ad71de8e8
[telegraf]  Step 4/7 : COPY telegraf.conf /etc/telegraf/telegraf.conf
[grafana]   Using cache
[grafana]    ---> f5058d98df9c
[grafana]   Step 4/8 : COPY ./*.sh /usr/src/app/
[telegraf]  Using cache
[telegraf]   ---> 5eda8aee0cf2
[telegraf]  Step 5/7 : COPY entry.sh /entry.sh
[grafana]   Using cache
[grafana]    ---> b00cb8dce5dd
[grafana]   Step 5/8 : RUN install_packages       fontconfig-config       fonts-dejavu-core       libfontconfig1       ucf       jq
[grafana]   Using cache
[grafana]    ---> 4cfdcde487c7
[grafana]   Step 6/8 : RUN chmod +x /usr/src/app/download.sh && /usr/src/app/download.sh "raspberrypi3"
[telegraf]  Using cache
[telegraf]   ---> 77fa467a394f
[telegraf]  Step 6/7 : RUN chmod +x /entry.sh
[telegraf]  Using cache
[telegraf]   ---> 2d21161e1e95
[telegraf]  Step 7/7 : CMD /entry.sh
[grafana]   Using cache
[grafana]    ---> 491fb63afc58
[grafana]   Step 7/8 : RUN dpkg -i /tmp/grafana.deb && rm /tmp/grafana.deb
[telegraf]  Using cache
[telegraf]   ---> c3ce6ce36529
[telegraf]  Successfully built c3ce6ce36529
[grafana]   Using cache
[grafana]    ---> d8d83533b2df
[grafana]   Step 8/8 : CMD /bin/sh /usr/src/app/entry.sh
[grafana]   Using cache
[grafana]    ---> baad4c6c2003
[grafana]   Successfully built baad4c6c2003
[sensor]     ---> 8789c9fc4e70
[sensor]    Step 2/16 : ARG BSEC_FILENAME=BSEC_1.4.7.3_Generic_Release_20190410.zip
[sensor]    Using cache
[sensor]     ---> d23c4f4af4f4
[sensor]    Step 3/16 : RUN install_packages         unzip
[sensor]    Using cache
[sensor]     ---> 06120d0a0717
[sensor]    Step 4/16 : WORKDIR /usr/src/app
[sensor]    Using cache
[sensor]     ---> 3d6b073f0c90
[sensor]    Step 5/16 : RUN git clone https://github.com/balena-io-playground/bsec_bme680_linux.git
[sensor]    Using cache
[sensor]     ---> dacda4ca8b34
[sensor]    Step 6/16 : RUN wget https://ae-bst.resource.bosch.com/media/_tech/media/bsec/$BSEC_FILENAME
[sensor]     ---> Running in cc87a55fd188
[sensor]    --2019-07-04 22:34:50--  https://ae-bst.resource.bosch.com/media/_tech/media/bsec/BSEC_1.4.7.3_Generic_Release_20190410.zip
[sensor]    
[sensor]    Resolving ae-bst.resource.bosch.com (ae-bst.resource.bosch.com)... 
[sensor]    139.15.248.75
[sensor]    Connecting to ae-bst.resource.bosch.com (ae-bst.resource.bosch.com)|139.15.248.75|:443... 
[sensor]    connected.
[sensor]    
[sensor]    HTTP request sent, awaiting response... 
[sensor]    404 Not Found
[sensor]    2019-07-04 22:34:51 ERROR 404: Not Found.
[sensor]    
[sensor]    Removing intermediate container cc87a55fd188
[Info]      Uploading images
[sensor]    The command '/bin/sh -c wget https://ae-bst.resource.bosch.com/media/_tech/media/bsec/$BSEC_FILENAME' returned a non-zero code: 8
[Success]   Successfully uploaded images
[Error]     Some services failed to build:
[Error]       Service: undefined
[Error]         Error: Information not available
[Error]       Service: undefined
[Error]         Error: Information not available
[Error]       Service: sensor
[Error]         Error: The command '/bin/sh -c wget https://ae-bst.resource.bosch.com/media/_tech/media/bsec/$BSEC_FILENAME' returned a non-zero code: 8
[Error]       Service: undefined
[Error]         Error: Information not available
[Error]     Not deploying release.
Remote build failed

```

escalateizzy, 2019-07-07:
Old version of BSEC has been removed. New one:
https://ae-bst.resource.bosch.com/media/_tech/media/bsec/BSEC_1.4.7.4_Generic_Release.zip

Using this, instead of 1.4.7.3, returns following error:
[Error]         Error: The command '/bin/sh -c ./make.sh' returned a non-zero code: 1

ValGor73, 2019-07-07:
+1
used new version
https://ae-bst.resource.bosch.com/media/_tech/media/bsec/BSEC_1.4.7.4_Generic_Release.zip

"The command '/bin/sh -c ./make.sh' returned a non-zero code: 1"
...
"Remote build failed"

DCanaan, 2019-07-07:
Hello there,
Same error here. I changed the following line in the dockerfile to match the new file name: 
`ARG BSEC_FILENAME=BSEC_1.4.7.4_Generic_Release.zip`

But I get the same error : 
`The command '/bin/sh -c ./make.sh' returned a non-zero code: 1`

Maybe there is still something else to modify...

[image]

ValGor73, 2019-07-08:
`Dockerfile.template	Bump BSEC version to latest 	**5 hours ago**`
It's OK!!!

Razzeee, 2019-07-08:
I'm still getting the same error, even after pulling or removing the complete source code.

chrisys, 2019-07-09:
@Razzeee can you try `balena push <appname> --nocache` ?

Razzeee, 2019-07-09:
Thanks, I got past the previous errror, it's still building right now. I will close this if everything works out fine.

closed 2019-07-09 by Razzeee