balena-labs-projects/balena-sense #27: bsec_bme680 returning nothing on Balena Fin

ampledata, 2019-07-09:
```i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 --                 ```

`/usr/src/app/bsec_bme680_linux/bsec_bme680` Returns Nothing.

Also tried ungrounding Addr and using 'secondary' to no avail.

chrisys, 2019-07-09:
@ampledata thanks for reporting; I'll run the project up on a Fin here and see what happens

chrisys, 2019-07-11:
@ampledata did you by chance update to v1.5.6 from a prior version? That would give the symptoms you're seeing. If so, we figured out that the state and config files from the previous version of the Bosch BSEC library are not compatible with the new one. You can delete these files from `/data/sensor` manually or update to v1.5.7 which should take care of it for you.

Otherwise, I've just run up a new deploy on a balenaFin v1.1 and everything looks good. If you're still having issues could you let me know what Fin you're using and what version of balenaSense and I'll try to reproduce.

chrisys, 2019-07-15:
Closing this - if you're still having an issue feel free to reopen with more information.

closed 2019-07-15 by chrisys