balena-labs-projects/balena-sense #30: at the end it goes in error

n1k0nx, 2019-07-13:
<Hi Ive received the sensor, and would install Balena Sense on my P0 W and P3 b+ 

all tests goes in the error below:

[Info]      Uploading images
[Success]   Successfully uploaded images
[Error]     Some services failed to build:
[Error]       Service: undefined
[Error]         Error: Information not available
[Error]       Service: grafana
[Error]         Error: The command '/bin/sh -c chmod +x /usr/src/app/download.sh && /usr/src/app/download.sh "raspberrypi3"' returned a non-zero code: 60
[Error]       Service: undefined
[Error]         Error: Information not available
[Error]       Service: telegraf
[Error]         Error: The command '/bin/sh -c curl -o /tmp/telegraf.deb https://dl.influxdata.com/telegraf/releases/telegraf_1.11.0-1_armhf.deb' returned a non-zero code: 60
[Error]     Not deploying release.
Remote build failed

Additional information may be available by setting a DEBUG=1 environment
variable: "set DEBUG=1" on a Windows command prompt, or "export DEBUG=1"
on Linux or macOS.

If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io

For bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/

coul you elp me please? 
thank you
Luca

chrisys, 2019-07-15:
Hey @n1k0nx this is fixed by version v1.5.8 released earlier today. 

The root cause of the problem is that the default Debian base image has been updated to Buster, which unfortunately brings [this issue](https://github.com/balena-io-library/base-images/issues/562) into play which affects the use of curl when it’s used to download the binaries for Grafana and Telegraf. We’ve updated the project to use wget now instead.

closed 2019-07-15 by chrisys

n1k0nx, 2019-07-15:
Hi Chrisys
I'm sorry but after update stil give me same issue. here with Debug=1

thank you in advance
luca 

Lucas-MacBook-Pro:balena-sense-master lucav$ debug=1
Lucas-MacBook-Pro:balena-sense-master lucav$ balena push balenasense

   ╭────────────────────────────────────────────────────────────────────╮
   │                                                                    │
   │                  Update available 11.6.0 → 11.7.0                  │
   │                                                                    │
   │   https://github.com/balena-io/balena-cli/blob/master/INSTALL.md   │
   │                                                                    │
   ╰────────────────────────────────────────────────────────────────────╯

[Info]      Starting build for balenasense, user lucaverr
[Info]      Dashboard link: https://dashboard.balena-cloud.com/apps/1493146/devices
[Info]      Building on arm03
[Info]      Pulling previous images for caching purposes...
[Success]   Successfully pulled cache images
[sensor]    Step 1/17 : FROM balenalib/raspberrypi3-python:3-build
[telegraf]  Step 1/7 : FROM balenalib/raspberrypi3
[grafana]   Step 1/8 : FROM balenalib/raspberrypi3
[influxdb]  Step 1/4 : FROM balenalib/raspberrypi3-alpine
[sensor]     ---> 7a6aae1a0243
[sensor]    Step 2/17 : ARG BSEC_FILENAME=BSEC_1.4.7.4_Generic_Release.zip
[sensor]    Using cache
[sensor]     ---> ca83e82f34a4
[sensor]    Step 3/17 : RUN install_packages         unzip
[sensor]    Using cache
[sensor]     ---> 83912551413d
[sensor]    Step 4/17 : WORKDIR /usr/src/app
[sensor]    Using cache
[sensor]     ---> e257a1a3ccb7
[sensor]    Step 5/17 : RUN git clone https://github.com/balena-io-playground/bsec_bme680_linux.git
[sensor]    Using cache
[sensor]     ---> db5a57f896ca
[sensor]    Step 6/17 : RUN wget https://ae-bst.resource.bosch.com/media/_tech/media/bsec/$BSEC_FILENAME
[grafana]    ---> 409a5ebb1eea
[grafana]   Step 2/8 : COPY ./grafana.ini /usr/share/grafana/conf/custom.ini
[sensor]    Using cache
[sensor]     ---> 4adbbf7fbe82
[sensor]    Step 7/17 : RUN unzip -d bsec_bme680_linux/src $BSEC_FILENAME
[telegraf]   ---> 409a5ebb1eea
[telegraf]  Step 2/7 : RUN curl -o /tmp/telegraf.deb https://dl.influxdata.com/telegraf/releases/telegraf_1.11.0-1_armhf.deb
[grafana]   Using cache
[grafana]    ---> 74936f483c00
[grafana]   Step 3/8 : COPY ./provisioning /usr/src/app/provisioning
[sensor]    Using cache
[sensor]     ---> 9b62cf06abf1
[sensor]    Step 8/17 : RUN echo $BSEC_FILENAME > version
[influxdb]   ---> 3a72b1214260
[influxdb]  Step 2/4 : RUN apk add influxdb
[influxdb]  Using cache
[influxdb]   ---> a487fe00be4b
[influxdb]  Step 3/4 : RUN sed -i 's|/var/lib/influxdb|/data/influxdb|g' /etc/influxdb/influxdb.conf
[grafana]   Using cache
[grafana]    ---> 010641f9e493
[grafana]   Step 4/8 : COPY ./*.sh /usr/src/app/
[sensor]    Using cache
[sensor]     ---> 8c2d16087eef
[sensor]    Step 9/17 : WORKDIR /usr/src/app/bsec_bme680_linux
[influxdb]  Using cache
[influxdb]   ---> 0a4ae68beaf7
[influxdb]  Step 4/4 : CMD influxd
[grafana]   Using cache
[grafana]    ---> 91f8725ab0aa
[grafana]   Step 5/8 : RUN install_packages       fontconfig-config       fonts-dejavu-core       libfontconfig1       ucf       jq
[... 85 lines cut ...]
[telegraf]  The command '/bin/sh -c curl -o /tmp/telegraf.deb https://dl.influxdata.com/telegraf/releases/telegraf_1.11.0-1_armhf.deb' returned a non-zero code: 60
[Success]   Successfully uploaded images
[Error]     Some services failed to build:
[Error]       Service: undefined
[Error]         Error: Information not available
[Error]       Service: grafana
[Error]         Error: The command '/bin/sh -c chmod +x /usr/src/app/download.sh && /usr/src/app/download.sh "raspberrypi3"' returned a non-zero code: 60
[Error]       Service: undefined
[Error]         Error: Information not available
[Error]       Service: telegraf
[Error]         Error: The command '/bin/sh -c curl -o /tmp/telegraf.deb https://dl.influxdata.com/telegraf/releases/telegraf_1.11.0-1_armhf.deb' returned a non-zero code: 60
[Error]     Not deploying release.
Remote build failed

Additional information may be available by setting a DEBUG=1 environment
variable: "set DEBUG=1" on a Windows command prompt, or "export DEBUG=1"
on Linux or macOS.

If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io

For bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/

Lucas-MacBook-Pro:balena-sense-master lucav$

chrisys, 2019-07-15:
@n1k0nx can you try with `--nocache` please?

n1k0nx, 2019-07-16:
@chrisys It work thank you from gniubbo progammer :)