balena-labs-projects/balena-sense #33: user_i2c_write: Remote I/O error

escalateizzy, 2019-08-08:
Hi all,

Has anyone has found this problem with sensor reading? 

user_i2c_write: Remote I/O error

It starts to happen randomly after some time with device up; and only solves if I reset device.

[image]

Thank you in advance

danielgreverose, 2019-09-11:
I have encountered this issue as well. Has happened twice in the last few days.

chrisys, 2019-10-02:
@escalateizzy @danielgreverose is this still happening for you guys? It would be great if you could share what hardware you're using; both what Pi and what sensors.

escalateizzy, 2019-10-04:
Hi, have just updated with last commit thanks. I will give you some feedback after a few days. Hardware using:
Raspberry Pi Zero WH
Generic BME680 sensor https://es.aliexpress.com/item/32961369966.html?spm=a2g0s.9042311.0.0.274263c0VqKpPP

Also tested in the past with Raspberry Pi 3, same error happened.

escalateizzy, 2019-10-05:
Unfortunately it happened again with latest commit (after 14 hours running).
[image]

escalateizzy, 2019-10-12:
Hi,
Disconnecting and connecting again without turning the Pi off, solved the issue. So really seems like a hardware problem. Will try to test with another sensor board.
Thanks!

chrisys, 2019-10-14:
@escalateizzy ahh at least we're starting to narrow it down. I'll close this for now then but feel free to reopen it if you're having the same trouble with a new sensor!

closed 2019-10-14 by chrisys