balena-labs-projects/balena-sense #34: Failed install on RPi3B+ 64

MadDenker, 2019-08-18:
I was unable to deploy the image on a Raspberry Pi 3B+ ARMv8 64-bit.  I believe the 64bit architecture is the issue.  Some error messages:

[image]

[image]

Can you provide a 64bit version?  Thanks.

MadDenker, 2019-08-22:
Looks like I might just need to point to different install deb packages.  For instance, I probably need to put telegraf_1.11.4-1_arm64.deb in this file: https://github.com/balena-io-projects/balena-sense/blob/a27b006148f51743c3ad564a09e811561beefb55/telegraf/Dockerfile.template

Any ideas on how best to do this?  Should I create a new branch or version or release?  Maybe a test for architecture before wgetting the package?

chrisys, 2019-08-26:
@MadDenker take a look at how the download for Grafana works - you can probably use %%BALENA_ARCH%% here and make a case for downloading the 64 bit version. A PR would be welcome once you get it up and running!

vschinazi, 2019-09-19:
@MadDenker did you ever manage to get this working? I am dying to get this working on a pi 4.

chrisys, 2019-10-01:
@MadDenker this is resolved in https://github.com/balena-io-projects/balena-sense/pull/40 - thanks for posting.

closed 2019-10-01 by chrisys

MadDenker, 2019-10-02:
Awesome!  Thanks!

MadDenker, 2019-12-28:
I am still having troubles installing it.  
[image]

The first error I see in the output is - 
 [telegraf]  debconf: delaying package configuration, since apt-utils is not installed