balena-labs-projects/balena-sense #35: Integration with Enviro + for Raspberry Pi

vschinazi, 2019-09-10:
Hi

Thank you for this fantastic project!
I am wondering if you have tested or have plans to integrate it with the Pimoroni Enviro+ sensor for Raspberry Pi. I am considering getting this sensor but also windering if   the Pimoroni BME 680 is the smarter choice. 

Thank you

chrisys, 2019-09-30:
@vschinazi just to keep you posted, myself and @shaunmulligan are planning to add this functionality in the next few weeks

vschinazi, 2019-09-30:
Hi

This is great news. I am really looking forward to it. I teach Environmental Psychology and plan to incorporate the Balena Sense project as part of my teaching.  I would love to tell you more about this. Not sure if this is the best thread ;)

Thank you

Victor

chrisys, 2019-10-01:
@vschinazi that sounds interesting - you can email me if you like! chriscw@balena.io

tradetoolswiki, 2019-12-17:
> @vschinazi just to keep you posted, myself and @shaunmulligan are planning to add this functionality in the next few weeks

Hi, I'm keen to get this working with an Enviro+ board I've got lying around. Has this support been added? I  couldn't find any other reference apart from this thread.

Thanks

chrisys, 2019-12-31:
@tradetoolswiki no work has been done on that yet, but if you have the hardware adding it to the project would be an appreciated contribution!

shaunmulligan, 2020-03-13:
closed by #59

shaunmulligan, 2020-03-13:
@vschinazi and @tradetoolswiki I added support for the enviro+ hat, currently doesn't support the PM sensor and the mic but has all the other basic sensors. I will try add the PM one as soon as I can. The microphone is a little bit more of a fiddle to get setup right.

closed 2020-03-13 by shaunmulligan

vschinazi, 2020-04-01:
> @vschinazi and @tradetoolswiki I added support for the enviro+ hat, currently doesn't support the PM sensor and the mic but has all the other basic sensors. I will try add the PM one as soon as I can. The microphone is a little bit more of a fiddle to get setup right.

Amazing! really looking forward to trying this. Thank you!

vschinazi, 2020-04-12:
Hi @shaunmulligan I just saw this project (https://github.com/tijmenvandenbrink/enviroplus_exporter#prerequisites) and looks like they managed the particle sensor. Might this be a good starting point to getting all the other sensors in Enviro+ in the Balena Sense project. Really looking forward to these additions. Thank you.

shaunmulligan, 2020-04-12:
@vschinazi thanks for pointing to that, I actually already have the PM sensor working and have it in a draft PR here: https://github.com/balenalabs/balena-sense/pull/69. The only sensor left out now is the microphone.

vschinazi, 2020-04-12:
Hi @shaunmulligan this is great news. Really looking forward to it! Thank you.