balena-labs-projects/balena-sense #47: Pi zero W missing from device list

vschinazi, 2019-10-13:
Hi

I wanted to run the Balena Sense project from my Pi Zero W but noticed there was not an option for selecting Pi Zero W in the device list. 

Can you please add this option? 

If not, I'm assuming I can download the Balena OS for the Pi Zero W from here (https://www.balena.io/os/#downloads-raspberrypi). However, I am not sure how to connect to my wifi once I download the image. 

Thank you.

chrisys, 2019-10-14:
Hi @vschinazi the device `Raspberry Pi (v1 or Zero)` also covers the Raspberry Pi Zero W, so you can go ahead and use that one and it'll work fine. I have created an issue internally to make this more clear in the future - thanks for reporting.

closed 2019-10-14 by chrisys

vschinazi, 2019-10-14:
Hi @chrisys thank you for the quick reply. 
I guess the best thing to do is to download the Balena OS for Pi (v1 or Zero) and then edit the wifi setup based on this: https://www.balena.io/docs/reference/OS/network/2.x/#wifi-setup

My comment was specific to the fact that Pi v1 or Pi Zero are not in the device list when adding a new device. 

I will try it this week and report back.

Thank you again for all your help.

chrisys, 2019-10-14:
@vschinazi you don't need to edit the WiFi setup manually; when you add a Pi Zero device you can still add the WiFi configuration at the point of download, just like the other devices, but if you have any trouble let us know!

vschinazi, 2019-10-14:
@chrisys This is exactly the issue that I am having. Since the Pi Zero is not listed in the devices I cannot at the WiFi configuration at the point of download. The device list only includes Pi2, Pi3 and Pi4. 

[image]

chrisys, 2019-10-14:
@vschinazi ahh I see the problem. You've added the application with the default device type of 'Raspberry Pi 3', because this is a different architecture, it won't let you add the Pi Zero to that app.

If you create another application with Raspberry Pi Zero as the default device type, it will then let you add Zeros and 3s to the same application.

Sorry for the confusion!

vschinazi, 2019-10-14:
Ahhhh :) Perfect. Thank you!
I really appreciate this!

k9gardner, 2022-03-14:
> If you create another application with Raspberry Pi Zero as the default device type, it will then let you add Zeros and 3s to the same application.

Ahh, maybe that's the same problem I'm having? I am using Balena Sound, and I originally set it up with a Pi 3. Does that mean that that's the default type? I see Device Type listed, but I don't see Default Device Type anywhere. I wanted to add a Zero W v1.1 to my application and the only Zero listed is the v2. Is it not possible to change the default on an existing application? If not, should I just scrap the whole application and redo it with the Zero as the default device type? Looking for the least painful way to get up and running with these two devices. Or... should I redo the whole thing with my pi 4 as the main device, due to the stuttering problems I've had with the 3? I'm rambling. Someone stop me.