balena-labs-projects/balena-sense #53: Project not building because `influxdb.conf` was relocated

tmigone, 2020-01-27:
Error when building with `balena push my-app`:
```
[influxdb]  sed: /etc/influxdb/influxdb.conf: No such file or directory
```

Upon inspecting the container it looks like `influxdb.conf` was moved from `/etc/influxdb/influxdb.conf` to `/etc/influxdb.conf`:

```
bash-5.0# ls -al /etc/influxdb/influxdb.conf
ls: /etc/influxdb/influxdb.conf: No such file or directory
bash-5.0# ls -al /etc/influxdb.conf 
-rw-r--r--    1 root     root         20855 Jan 27 13:47 /etc/influxdb.conf
```

chrisys, 2020-01-27:
Related:
https://pkgs.alpinelinux.org/contents?branch=edge&name=influxdb&arch=armv7&repo=community

and
https://pkgs.alpinelinux.org/contents?file=&path=&name=influxdb&branch=v3.10&repo=community&arch=armv7

This shows the path change for the configuration file.

tmigone, 2020-01-27:
Fixed in #54

closed 2020-01-27 by tmigone