balena-labs-projects/balena-sense #63: Raspberry is crashing

Gizimo, 2020-02-19:
Hi,
i have implemented the Project on a Raspberry Zero/W
https://www.balena.io/blog/build-an-environment-and-air-quality-monitor-with-raspberry-pi/
Now my Rasp is constantly crashing after few days of working perfect

Got this error:

`19.02.20 13:14:22 (+0100)  sensor  ----------------------------------------
19.02.20 13:14:22 (+0100)  sensor  Exception happened during processing of request from ('172.17.0.4', 50622)
19.02.20 13:14:22 (+0100)  sensor  Traceback (most recent call last):
19.02.20 13:14:22 (+0100)  sensor    File "/usr/local/lib/python3.5/socketserver.py", line 313, in _handle_request_noblock
19.02.20 13:14:22 (+0100)  sensor      self.process_request(request, client_address)
19.02.20 13:14:22 (+0100)  sensor    File "/usr/local/lib/python3.5/socketserver.py", line 341, in process_request
19.02.20 13:14:22 (+0100)  sensor      self.finish_request(request, client_address)
19.02.20 13:14:22 (+0100)  sensor    File "/usr/local/lib/python3.5/socketserver.py", line 354, in finish_request
19.02.20 13:14:22 (+0100)  sensor      self.RequestHandlerClass(request, client_address, self)
19.02.20 13:14:22 (+0100)  sensor    File "/usr/local/lib/python3.5/socketserver.py", line 681, in __init__
19.02.20 13:14:22 (+0100)  sensor      self.handle()
19.02.20 13:14:22 (+0100)  sensor    File "/usr/local/lib/python3.5/http/server.py", line 422, in handle
19.02.20 13:14:22 (+0100)  sensor      self.handle_one_request()
19.02.20 13:14:22 (+0100)  sensor    File "/usr/local/lib/python3.5/http/server.py", line 410, in handle_one_request
19.02.20 13:14:22 (+0100)  sensor      method()
19.02.20 13:14:22 (+0100)  sensor    File "/usr/src/app/scripts/sensor.py", line 112, in do_GET
19.02.20 13:14:22 (+0100)  sensor      self.wfile.write(json.dumps(measurements[0]['fields']).encode('UTF-8'))
19.02.20 13:14:22 (+0100)  sensor    File "/usr/local/lib/python3.5/socket.py", line 594, in write
19.02.20 13:14:22 (+0100)  sensor      return self._sock.send(b)
19.02.20 13:14:22 (+0100)  sensor  BrokenPipeError: [Errno 32] Broken pipe
19.02.20 13:14:22 (+0100)  sensor  ----------------------------------------`

robertgzr, 2020-06-02:
I just posted a reply with the results of our investigation here: https://forums.balena.io/t/startup-and-shutdown-problems-balena-sense/119345/32

tl;dr see https://github.com/balena-os/meta-balena/issues/1910#issuecomment-637605110

saintaardvark, 2020-09-03:
Hi @Gizimo -- one suggestion is to try upgrading the OS for your Pi Zero.  The latest version in production (2.54.2+rev1 as I write this) enables support for zram, which may help performance in cases where resources are constrained.  Can you give this a try and see how it works?