balena-labs-projects/balena-sense #70: V 1.9.0 not working with BME680 and RPI3

fuzzybear62, 2020-03-15:
The container "sensors" aborts with the error:
...
15.03.20 15:27:30 (+0100)  sensor    File "/usr/src/app/scripts/sensor.py", line 118, in do_GET
15.03.20 15:27:30 (+0100)  sensor      measurements = balenasense.sample()
15.03.20 15:27:30 (+0100)  sensor    File "/usr/src/app/scripts/sensor.py", line 90, in sample
15.03.20 15:27:30 (+0100)  sensor      return self.apply_offsets(self.sensor.get_readings(self.sensor))
15.03.20 15:27:30 (+0100)  sensor  AttributeError: 'balenaSense' object has no attribute 'sensor'
15.03.20 15:27:30 (+0100)  sensor  ----------------------------------------

roman-mazur, 2020-03-19:
Can be related to this change:
https://github.com/balenalabs/balena-sense/commit/11949fbaccd2a139325ad721ff97f6429abbb78b#diff-b0076544b1cb64682ccb727d8301d050L52

chrisys, 2020-03-19:
There's a proposed fix for this in this branch: https://github.com/balenalabs/balena-sense/tree/chrisys-v190-patch

We're just testing with the other sensor types before merging

chrisys, 2020-03-19:
@fuzzybear62 thanks for taking the time to report this issue, it should now be fixed in the latest version v1.9.1.

closed 2020-03-19 by chrisys

TH2SPA, 2020-03-19:
@chrisys Great! Thank you

PR #71 (2020-03-19): Reinstate BME680 object load
Fixes #70