balena-labs-projects/balena-sense #75: Sensor isn't reading data properly.

sirdir, 2020-05-08:
My setup Raspberry Pi 3 B+ (balenaOS 2.47.0+rev1 + this repo tag v1.9.2)
also tried with (Raspberry Pi 2 B and Raspberry Pi Zero W)

I'm using **Unbranded BME680 breakout (CJMCU-680)** as sensor. I have spare sensor that was  sealed inside antistatic bag. Tried with both, not working. So sensor also in't a problem.

And year ago it worked completelly fine. If I remember correcly I'm used balenaOS 2.29.0+rev1. 

But today I try to revive my Pi fo home usage and I recieve this error in terminal. Apart from software there was no changes.

```
08.05.20 03:19:30 (+0200)  sensor  ----------------------------------------
08.05.20 03:19:32 (+0200)  sensor  user_i2c_write: Remote I/O error
08.05.20 03:19:34 (+0200)  sensor  Capture thread exited; restarting
08.05.20 03:19:39 (+0200)  sensor  user_i2c_write: Remote I/O error
08.05.20 03:19:40 (+0200)  sensor  172.18.0.4 - - [08/May/2020 01:19:40] "GET / HTTP/1.1" 200 -
08.05.20 03:19:40 (+0200)  sensor  ----------------------------------------
08.05.20 03:19:40 (+0200)  sensor  Exception happened during processing of request from ('172.18.0.4', 47992)
08.05.20 03:19:40 (+0200)  sensor  Traceback (most recent call last):
08.05.20 03:19:40 (+0200)  sensor    File "/usr/local/lib/python3.5/socketserver.py", line 313, in _handle_request_noblock
08.05.20 03:19:40 (+0200)  sensor      self.process_request(request, client_address)
08.05.20 03:19:40 (+0200)  sensor    File "/usr/local/lib/python3.5/socketserver.py", line 341, in process_request
08.05.20 03:19:40 (+0200)  sensor      self.finish_request(request, client_address)
08.05.20 03:19:40 (+0200)  sensor    File "/usr/local/lib/python3.5/socketserver.py", line 354, in finish_request
08.05.20 03:19:40 (+0200)  sensor      self.RequestHandlerClass(request, client_address, self)
08.05.20 03:19:40 (+0200)  sensor    File "/usr/local/lib/python3.5/socketserver.py", line 681, in __init__
08.05.20 03:19:40 (+0200)  sensor      self.handle()
08.05.20 03:19:40 (+0200)  sensor    File "/usr/local/lib/python3.5/http/server.py", line 422, in handle
08.05.20 03:19:40 (+0200)  sensor      self.handle_one_request()
08.05.20 03:19:40 (+0200)  sensor    File "/usr/local/lib/python3.5/http/server.py", line 410, in handle_one_request
08.05.20 03:19:40 (+0200)  sensor      method()
08.05.20 03:19:40 (+0200)  sensor    File "/usr/src/app/scripts/sensor.py", line 123, in do_GET
08.05.20 03:19:40 (+0200)  sensor      measurements = balenasense.sample()
08.05.20 03:19:40 (+0200)  sensor    File "/usr/src/app/scripts/sensor.py", line 95, in sample
08.05.20 03:19:40 (+0200)  sensor      return self.apply_offsets(self.sensor.get_readings(self.sensor))
08.05.20 03:19:40 (+0200)  sensor    File "/usr/src/app/scripts/bme680.py", line 45, in get_readings
08.05.20 03:19:40 (+0200)  sensor      'temperature': float(self.data['temperature']),
08.05.20 03:19:40 (+0200)  sensor  TypeError: 'NoneType' object is not subscriptable
08.05.20 03:19:40 (+0200)  sensor  ----------------------------------------
08.05.20 03:19:41 (+0200)  sensor  Capture thread exited; restarting
08.05.20 03:19:46 (+0200)  sensor  user_i2c_write: Remote I/O error
08.05.20 03:19:48 (+0200)  sensor  Capture thread exited; restarting
08.05.20 03:19:50 (+0200)  sensor  172.18.0.4 - - [08/May/2020 01:19:50] "GET / HTTP/1.1" 200 -
08.05.20 03:19:50 (+0200)  sensor  ----------------------------------------
```

Any help?

Bakytbekz, 2020-05-27:
I have the same problem. ((

sirdir, 2020-05-27:
> I have the same problem. ((

```
git clone git@github.com:balenalabs/balena-sense.git
git checkout v1.8.3
```
It will solve your problem, but would be nice if maintainers (or someone with python knowledge) fix this problem in master.

sirdir, 2020-06-25:
@chrisys why you don't mark it as a bug?