balena-labs-projects/balena-sense #77: Including the BME280

m0roj, 2020-05-18:
It would be helpful if the sense module BME280 was included within balena-sense for those not interested in having VOC gas sensing provided for in the BME680. They are both compatible in all other respects as far as I can see. My log for using the BME280 produced:
This is the log:
16.05.20 20:11:36 (+0000) sensor Enviro Plus hat not found
16.05.20 20:11:36 (+0000) sensor BME680 not found on 0x76, trying 0x77
16.05.20 20:11:36 (+0000) sensor BME680 not found on 0x77
16.05.20 20:11:36 (+0000) sensor Sense HAT not found
16.05.20 20:11:37 (+0000) sensor 1-wire sensor not found
16.05.20 20:11:37 (+0000) sensor No suitable sensors found! Exiting.
16.05.20 20:11:40 (+0000) telegraf 2020-05-16T20:11:40Z E! [inputs.http] Error in plugin: [url=http://sensor]: Get http://sensor: dial tcp 172.17.0.5:80: connect: connection refused
16.05.20 20:11:42 (+0000) Service exited ‘sensor sha256:1e449bd09f641af109eef47e0a13763614650d81934ed05365a328acac3150e9’
17.05.20 16:52:26 (+0000) Supervisor starting
Any thoughts/solutions please?

GOTO-GOSUB, 2020-12-29:
+1 from me, I would also like to see support for the BME280 but am not savvy enough to add the required library. In my case I am using this particular device:

https://shop.pimoroni.com/products/bme280-breakout

alanb128, 2021-06-13:
The BME280 is now supported in version 2 of balenaSense. Please check the project's readme before upgrading!

closed 2021-06-13 by alanb128