balena-labs-projects/balena-sense #98: [Question] Multiple Sensors, One Device

mbauer599, 2021-04-18:
This project is almost perfect for my use. For my purpose it makes sense to have multiple sensors attached to one device (One in water temp sensor and one air temp sensor). Is this possible to use multiple sensors like this? Ideally they could be different devices as well, (IE: BME680 + DS18B20 or any number of these in combination to monitor different metrics). 

Correct me if i'm wrong but it looks like when multiple devices are referenced in config doc it's referencing multiple different devices each with one sensor attached when centrally aggregating this data? If this is not the case then I would happily take any documentation or examples that would clarify this. If so, then is this possible from within this or am I going to hax it together?

kekkers23, 2021-05-08:
Seems this project is a bit dead, any chance you have a copy of the Bosch BSEC 1.4.8.0 generic release as its no longer on their website?

alanb128, 2021-06-09:
With the recent release of balenaSense V2, you can now connect multiple sensors. Please see the project's README for more details.

closed 2021-06-09 by alanb128