balena-labs-projects/balena-sense #104: Orange pi zero install problem

lasuridze, 2021-06-06:
Error in plugin: [url=http://sensor:7575]: Get "http://sensor:7575": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

[sensor_bitter-resonance-06.06.21_17_14_43_(+0400).txt](https://github.com/balenalabs/balena-sense/files/6604247/sensor_bitter-resonance-06.06.21_17_14_43_.%2B0400.txt)

alanb128, 2021-06-07:
It looks like the Orange Pi Zero does not use the hard-coded I2C bus number 1. I'm testing an addition to the sensor block so that you can specify a different bus. I'll post more information here when that is available.

alanb128, 2021-06-09:
@lasuridze can you try pulling the latest version of the sensor block and setting the `BUS_NUMBER` service variable to `0` (or whichever bus you are using?)