balena-labs-projects/balena-sense #107: BME680 Now working out of the box

mbauer599, 2021-06-26:
I'm having the following issue, multiple BME680 sensors are failing to work with v2 out of the box -- sensors that were (and are) working with v1. I'm using devices that are being migrated from a v1 project, so this could be the issue. Any help on how to do this properly is appreciated. 

Looks like sensor script is failing and hanging:
```
Starting mqtt client, publishing to mqtt:1883
Using MQTT publish interval: 8.0 sec(s)
Initializing sensors...
======== Searching i2c bus for devices... ========
Found device at 0x77
======== Removing existing devices from the i2c bus... ========
Deleting device found at 0x77.
======== Unloading any existing modules... ========
New active: [119]
======== Loading devices found... ========
modprobe: FATAL: Module bme680-i2c not found in directory /lib/modules/4.19.75
Loading device bme680-i2c (chip ID 97) on address 0x77.
Loaded 1 of 1 device(s) found
initializing reading
IIO context created: local
Backend version: 0.16 (git tag: v0.16)
Backend description string: Linux 1b23cdd24eb2 4.19.75 #1 SMP Mon Mar 23 10:27:22 UTC 2020 armv7l
IIO context has 1 attributes: 
        local,kernel: 4.19.75
IIO context has 0 devices:
```

I will stress that it's working on V1 so I don't think hardware is the issue. As well, I have updated to SUPERVISOR VERSION
12.8.8 from 10.x and have the issue on both. 

V1 Output from Sensor:
```
root@fec2c46:/# curl sensor
{"temperature": 22.62, "pressure": 974.29, "humidity": 46.82, "air_quality_score": 25.0, "air_quality_score_accuracy": 0, "eco2_ppm": 0.0, "bvoce_ppm": 0.0}
```

alanb128, 2021-07-01:
What type of device are you using?

mbauer599, 2021-07-01:
Multiple BME680s. If you are looking for manufacturers:
* [Adafruit](https://www.adafruit.com/product/3660)
* [KOOBOOK](https://www.amazon.com/gp/product/B07TV9QZQX)
* [ACEIRMC](https://www.amazon.com/ACEIRMC-Temperature-Humidity-Compatible-Raspberry/dp/B08FC4MNJC/)

I also am using a [Sense HAT](https://www.amazon.com/gp/product/B014HDG74S) that also does not appear to work, however I did not (and have not) tested this on V1 so I'm excluding this for now.

alanb128, 2021-07-01:
Sense V2 won't work on the Pi Zero because Influx does not have an ARMv6 build, but there should not be an issue with the Pi2 or 3. What version of balenaOS are you running?

mbauer599, 2021-07-01:
In the long run that's not an issue for me, since I'm using a centralized Influx/Grafana for all my sensor data (one view for many rooms in my house), although I haven't even thought about building this for V2 of this project yet. Need to get the sensors working first. 

Copied from my PI 2:
SUPERVISOR VERSION -- 12.8.8
HOST OS VERSION -- balenaOS 2.48.0+rev1

I'm also running: balenaOS 2.54.2+rev1

Would updating this version be of use for testing or functionality?

benhowes, 2022-06-07:
FWIW, I was hitting this error and updating the host OS (`2.98.33` after the update) worked for me. I believe this is due to the use of the kernel modules for IO in v2

mbauer599, 2022-06-07:
As of the time of this error occuring, updating the os did nothing. I did
try running on multiple pi's as well (multiple versions of pi too,
including a 2/3/4) to no avail. I tried upgrading and downgrading os to no
avail. I actually just gave up and rolled my own since my target was pi
zeros anyway, pushing to cloud influxdb and grafana cloud so I don't need
any more power on the pi.