balena-labs-projects/balena-sense #112: Fin team reports that sensors are not working out of the box on the Fin

chrisys, 2021-07-08:
@alanb128 the guys mentioned this to me in the hardware call but I've not tested yet, so just logging it here.

chrisys, 2021-07-08:
I've just tested joining the open fleet with a Fin and a BME680 - seems to work OK.

alanb128, 2021-07-08:
I just tested the simple-sensor with a BMP-280 on the Fin and it worked as well. I did not see the BME680 kernel module in balenaOS (2.80.3+rev1, prod) so I'm surprised that one worked @chrisys - which version of the OS did you use?

chrisys, 2021-07-08:
@alanb128 my device is in the open fleet if you want to take a look, it's balenaOS 2.80.3+rev1 and has the bme680 modules loaded!

alanb128, 2021-07-08:
I see that kernel module now, the manifest explorer dropdown must have been out of sync.