balena-labs-projects/balena-sense #119: multiple bme680 in diferents rooms?

ser6iogl92, 2021-12-01:
Hello, i am trying to put diferents bme680 in only one raspberry.
when i put 2 only appears 1. how i can add differents? i should change name on transformers.py? how i do it? 
thanks so much. i am learning step by step.

alanb128, 2021-12-27:
You'll need to change the I2C address of one of the bme680s if you attach more than one to the same Pi. If you are using the Adafruit version, "By default, the i2c address is 0x77.  If you add a jumper from SDO to GND, the address will change to 0x76."
A better way to have multiple sensors in different rooms might be to use the Simple Sensor project with balenaSense as described in our blog article here: https://www.balena.io/blog/balenasense-v2-updated-temperature-pressure-and-humidity-monitoring-for-raspberry-pi/#aggregate-your-data-from-multiple-devices-and-sensors

nmctiglao-upd, 2022-02-27:
> You'll need to change the I2C address of one of the bme680s if you attach more than one to the same Pi. If you are using the Adafruit version, "By default, the i2c address is 0x77. If you add a [...]

Hi Alan, how can I add a container running Home Assistant to this container?